adafruit/Adafruit_AHRS #5: Rotating bunny uses deprecated libraries

Nathnainiel, 2019-04-21:
The processing sketch for visualising the rotation from the IMU is incompatible with Processing 3 and, more importantly, uses the OBJLoader library that is no longer available. Adafruit_BNO055 repo has included OBJLoader in the repo, perhaps that is an alternative to rewriting the processing sketch with different libraries (https://github.com/adafruit/Adafruit_BNO055

ladyada, 2019-04-21:
we don't have any time for porting to proc3, but it would be really helpful - do you want to take a stab at it?

ladyada, 2020-04-19:
hi we've created a web-based visualizer!
https://learn.adafruit.com/how-to-fuse-motion-sensor-data-into-ahrs-orientation-euler-quaternions

closed 2020-04-19 by ladyada