adafruit/Adafruit_AHRS #6: Update code to work with the LSM9DS1

lcipolina, 2019-12-09:
The code for AHRS should be updated to be able to do sensor fusion work with the newest version of the chip. This is the code to change
https://github.com/adafruit/Adafruit_AHRS/blob/master/examples/ahrs_fusion_usb/ahrs_fusion_usb.ino

ladyada, 2019-12-09:
yeah this library needs a big cleanup. is that something you'd like to PR? :)

lcipolina, 2019-12-09:
Thank you for replying. I can take a look, yes.
 First the magnetometer script  needs to be updated (to do the mag calibration)
Then  this code should be further adapted for the S1

ladyada, 2019-12-09:
thank you it would be most appreciated :)

outlandnish, 2019-12-11:
@ladyada @lcipolina I've got a version of the code running with a modified version of the SparkFun LSM9DS1 library. Is it alright if I PR that version or would you prefer that any PR uses the Adafruit LSM9DS1 library that uses the Adafruit Sensors interface?

ladyada, 2019-12-11:
either/both ways appreciated :)

outlandnish, 2019-12-16:
Here's the calibration code using [my fork of the SparkFun LSM9DS1 library](https://github.com/intentfulmotion/lib-driver-lsm9ds1). However, I can't get it to calibrate properly with my own hardware. If this code works for you, please let me know!

```c++
#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>

#define ST_LSM303DLHC_L3GD20        (0)
#define ST_LSM9DS1                  (1)
#define NXP_FXOS8700_FXAS21002      (2)

// Define your target sensor(s) here based on the list above!
// #define AHRS_VARIANT    ST_LSM303DLHC_L3GD20
#define AHRS_VARIANT   ST_LSM9DS1

// Include appropriate sensor driver(s)
#if AHRS_VARIANT == ST_LSM303DLHC_L3GD20
#include <Adafruit_L3GD20_U.h>
#include <Adafruit_LSM303_U.h>
#elif AHRS_VARIANT == ST_LSM9DS1
#include <SparkFunLSM9DS1.h>
#define AG_ADDRESS  0x6B
#define MAG_ADDRESS 0x1E
#elif AHRS_VARIANT == NXP_FXOS8700_FXAS21002
#include <Adafruit_FXAS21002C.h>
#include <Adafruit_FXOS8700.h>
#else
#error "AHRS_VARIANT undefined! Please select a target sensor combination!"
#endif

// Create sensor instances.
#if AHRS_VARIANT == ST_LSM303DLHC_L3GD20
Adafruit_L3GD20_Unified       gyro(20);
Adafruit_LSM303_Accel_Unified accel(30301);
Adafruit_LSM303_Mag_Unified   mag(30302);
#elif AHRS_VARIANT == ST_LSM9DS1
LSM9DS1 imu;
int16_t ax, ay, az, gx, gy, gz, mx, my, mz;
#elif AHRS_VARIANT == NXP_FXOS8700_FXAS21002
Adafruit_FXAS21002C gyro = Adafruit_FXAS21002C(0x0021002C);
Adafruit_FXOS8700 accelmag = Adafruit_FXOS8700(0x8700A, 0x8700B);
#endif

// NOTE: THIS IS A WORK IN PROGRESS!

// This sketch can be used to output raw sensor data in a format that
// can be understoof by MotionCal from PJRC. Download the application
// from http://www.pjrc.com/store/prop_shield.html and make note of the
// magentic offsets after rotating the sensors sufficiently.
//
// You should end up with 3 offsets for X/Y/Z, which are displayed
// in the top-right corner of the application.

void setup()
{
  Serial.begin(115200);

  // Wait for the Serial Monitor to open (comment out to run without Serial Monitor)
  // while(!Serial);

  Serial.println(F("Adafruit nDOF AHRS Calibration Example")); Serial.println("");

#if AHRS_VARIANT == NXP_FXOS8700_FXAS21002 || AHRS_VARIANT == ST_LSM303DLHC_L3GD20
  // Initialize the sensors.
  if(!gyro.begin())
  {
    /* There was a problem detecting the gyro ... check your connections */
    Serial.println("Ooops, no gyro detected ... Check your wiring!");
    while(1);
  }
[... 117 lines cut ...]
    Serial.println();
#endif
#if AHRS_VARIANT == ST_LSM9DS1
    Serial.print(ax);
    Serial.print(",");
    Serial.print(ay);
    Serial.print(",");
    Serial.print(az);
    Serial.print(",");
    Serial.print(gx);
    Serial.print(",");
    Serial.print(gy);
    Serial.print(",");
    Serial.print(gz);
    Serial.print(",");
    Serial.print(mx);
    Serial.print(",");
    Serial.print(my);
    Serial.print(",");
    Serial.print(mz);
    Serial.println();
#endif

  delay(50);
}

ladyada, 2020-02-02:
refactored library, LSM9DS1 is in there now

closed 2020-02-02 by ladyada

LazaroFilm, 2022-12-24:
[This example](https://github.com/adafruit/Adafruit_AHRS/blob/69b3a854be83d8b6c6d897d801882a222d1ae510/examples/ahrs_fusion_ble_nrf51/ahrs_fusion_ble_nrf51.ino) still says `to do` in the code under the LSM9DS1 on line 25.