adafruit/Adafruit_AHRS #10: Adafruit_AHRS_NXPFusion can give you absolute heading

moritzmaier, 2020-04-14:
Hi, I've stumbled upon this line of code in ```Adafruit_AHRS_NXPFusion.cpp``` today
if (fabsf(mx) >= 20.0f && fabsf(mx) >= 20.0f && fabsf(mx) >= 20.0f)

This line has a huge effect. Almost certainly ValidMagCal will be zero and the first orientation lock is not performed. That does not matter, if you expect yaw to be zero initially. But the Kalman filter can give you absolute heading, i.e. the compass orientation where heading = 0 means you head towards magnetic north.

I changed my code to

// *********************************************************************************
  // initial orientation lock to accelerometer and magnetometer eCompass
  // orientation
  // *********************************************************************************
  //if (fabsf(mx) >= 20.0f && fabsf(mx) >= 20.0f && fabsf(mx) >= 20.0f) {
  //  ValidMagCal = 1;
  //} else {
  //  ValidMagCal = 0;
  //}

  ValidMagCal = 1;

  // do a once-only orientation lock after the first valid magnetic calibration
  if (ValidMagCal && !FirstOrientationLock) {
    // get the 6DOF orientation matrix and initial inclination angle
    feCompassNED(RPl, &DeltaPl, Mag, Accel);

    // get the orientation quaternion from the orientation matrix
    fQuaternionFromRotationMatrix(RPl, &qPl);

    // set the orientation lock flag so this initial alignment is only performed
    // once
    FirstOrientationLock = 1;
    printf("Performed initial orientation lock\n");
  }

Now I obtain absolute heading, instead of relative yaw with respect to the initial orientation. I still experience some drift, but I believe that this is a matter of magnetometer calibration and timing (I am using a Raspberry Pi 4).

Note that on a calibrated magnetometer mx, my, and mz are between -50 and 50 the area where the abs() of mx (or of all three) is greater than 20 is just a random direction (or a random portion of the sphere with radius 50).

See
https://de.mathworks.com/help/nav/ug/magnetometer-calibration.html
https://www.pjrc.com/store/prop_shield.html

ladyada, 2020-04-14:
huh- good idea! ok can you submit a PR and we'll try it out :)

moritzmaier, 2020-04-14:
> huh- good idea! ok can you submit a PR and we'll try it out :)

Done. But clang check fails. 

I don't know if you really want absolute heading all the time with NXP, or if you want to be consistent with Mahony and Madgwick AHRS.

commit 007b6f91 (2020-04-16): Merge pull request #11 from moritzmaier/master

Update Adafruit_AHRS_NXPFusion.cpp resolves #10
closed 2020-04-16 by siddacious