adafruit/Adafruit_AHRS #13: LSM9DS1

insidecircuits, 2020-07-24:
- Arduino board:  Uno (m328)
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.11

Using the "calibrated_orientation.ino" example file and uncommenting the lines to use the LSM9DS1 sensor does not output the good data.
The calculations are done but the orientation angles drift.

ladyada, 2020-07-24:
there's always going to be drift, this is normal

closed 2020-07-24 by ladyada

insidecircuits, 2020-07-25:
In fact it is not about filter drift, I will try to explain better.
Each IMU contains 3 sensors, and each manufacturer orients these 3 sensors in different ways to each other, it can be seen here, the difference between LSM9DS0 and LSM9DS1
[image]
[image]
Is this handled somwhere in the Unified Sensor library? I did not find a setting for this.

What I had to do for the LSM9DS1 was to rearange the axis manually in the filter calculation function, but I could not figure out how to rotate the magnetometer:
`filter.update(gy, -gx, gz, -accel.acceleration.y, accel.acceleration.x, accel.acceleration.z, 0, 0, 0);`

Please take a look into this, it is a big issue.

ladyada, 2020-07-25:
magnetic data is here
https://github.com/adafruit/Adafruit_AHRS/blob/master/examples/calibrated_orientation/calibrated_orientation.ino#L94

insidecircuits, 2020-07-25:
Yes, that I am aware of.
Just to make sure that you understand the modifications I had to make:
This is your original code:
`filter.update(gx, gy, gz, accel.acceleration.x, accel.acceleration.y, accel.acceleration.z, mag.magnetic.x, mag.magnetic.y, mag.magnetic.z);`

This is my modification for Gyroscope and Accelerometer only.
`filter.update(gy, -gx, gz, -accel.acceleration.y, accel.acceleration.x, accel.acceleration.z, 0, 0, 0);`

The magnetometer is "0, 0, 0" because I could not find a working solution, the default "mag.magnetic.x, mag.magnetic.y, mag.magnetic.z" is not giving good results, as it's the case for accelerometer and gyro also.

The sensors have to be rotated around in order for the LSM9DS1 sensor to be compatible with your library. How to rotate around  the magnetometer, this is the question.

ladyada, 2020-07-25:
you could swap around the values and/or add negatives