adafruit/Adafruit_AHRS #21: wrong unit documentation?

jerabaul29, 2021-05-14:
I wonder if the unit documentation here is wrong:

https://github.com/adafruit/Adafruit_AHRS/blob/1425f14fbd2312bc57bcb0bd9bd9372372eaf363/src/Adafruit_AHRS_FusionInterface.h#L54-L56

Are you sure the unit of the acceleration is in gs? Is is not in m/s**2 rather than gs?

MarcBousquet, 2021-07-06:
Hello,
I would believe there is no consequences of using one or the other unit because it is a linear acceleration. 
The only difference between g and m/s^2 is a factor of ~9.8 (1g = 9.8 m/s^2), so using g would be, to some extend, the same as using km/s^2 with a 1/1000  factor instead of 9.8

-M-

jerabaul29, 2021-07-06:
I would agree if only accelerations were used in the processing, but I wonder if it makes a difference when all data are fused with the Kalman filter. For example, if there is centrifugal acceleration, this creates a relation between acceleration / angular rate of rotation / velocity and units may start to play a role, right? Not sure exactly how the Kalman filter is formulated here though.

MarcBousquet, 2021-07-06:
The angular rate of rotation is in deg/s and *must* be because this is not linear. the Update() function first change it to rad/s. 
Even if there is a constant factor between deg and rad, both units cannot be swapped because it is not linear.

jerabaul29, 2021-07-06:
Yes, I agree, but this is not really my point :) . I may be confused though, and I think it would be necessary to look at exactly how the Kalman filter is formulated to know some more.

MarcBousquet, 2021-07-06:
I was about to explain with quite a complex example but, reading at the code, i found the ultimate answer : the acceleration vector is normalized so any constant factor disappears.

lets consider your IMU is stall flat. Your IMU measures an acc of [0, 0, 9.8] m/s^2.
First the function normalize the vector so it does :
`recipNorm = invSqrt(ax * ax + ay * ay + az * az)` ==> InvSqrt(0\*0 + 0\*0 + 9.8\*9.8) = 1/Sqrt(9.8\*9.8) = 1 / 9.8
`ax *= recipNorm` ==> ax = 0 * 1 / 9.8 = 0
`ay *= recipNorm` ==> ay = 0 * 1 / 9.8 = 0
`az *= recipNorm` ==> az = 9.8 * 1 / 9.8 = 1
so Acc measurement is now [0, 0, 1] exactly as if you used g instead of m/s^2. And you could even use km/s^2 or mm/s^2 it would do the same.

jerabaul29, 2021-07-06:
Ok, thanks!

I was a bit unsure of this because I am pretty sure that there are some (possibly more advanced) formulations of Kalman filters that also hold an estimate for the velocity and take into account relations between velocity and acceleration (such as centrifugal effects). But I guess this is only for Kalman filters with extra input such as velocity from a GPS. Or maybe I just remember some old lectures wrongly :) .

jerabaul29, 2021-07-06:
(and in this case when different quantities get coupled, I think units would matter).

MarcBousquet, 2021-07-06:
> Ok, thanks!
>
>
>
> I was a bit unsure of this because I am pretty sure that there are some (possibly more advanced) formulations of Kalman filters that also hold an estimate for the velocity and take [...]

You're right! Unit does matter when coupling the acceleration measurements to positioning system and velocity computation.

jerabaul29, 2021-07-06:
😊👍

closed 2021-07-06 by jerabaul29