adafruit/Adafruit_AHRS #25: Get positions?

Fraubned, 2021-12-02:
Hi there, 
I wanted to get positions from linear accelerations by used NXP /Kalman algorithm. But I got very big faults even zero positions. How can I fix it? Have you any suggestions about them? (I am getting positions from v = v0 + a.t, p = p0 + v.t)

VA7NFH, 2021-12-02:
Can you post your full code? You have to integrate it twice: v += v0 * dt, p += p0 * dt.

Fraubned, 2021-12-02:
float x,y,z; 
fusion.GetLinearAcceleration(&x,&y,&z);
v0x=0;
v0y=0;
v0z=0;
pos0x = 0;
pos0y = 0;
pos0z = 0;
in loop: 
dt = 0.005 //1/200hz
v0x = v0x + x*dt
v0y = v0y + y*dt
v0z = v0z + z*dt
pos0x = pos0x + v0x*dt
pos0y = pos0y + v0y*dt
pos0z = pos0z + v0z*dt

here my zero_movement values :
pos0x = -2.42397
pos0y = -19.3137
pos0z = -1108.85

I collected measured data from Xsens 9 axis Imu sensor for 30 seconds

VA7NFH, 2021-12-02:
You need to move fusion.GetLinearAcceleration(&x, &y, &z) into the loop, and you should be using a dynamic dt instead of a constant if you want better accuracy.

Fraubned, 2021-12-04:
İt is already in a loop. But I cannot get never. Do you have another suggestions?

VA7NFH, 2021-12-04:
Can you post the entire source code?

Fraubned, 2021-12-05:
Sorry I cannot share source codes online, It is not personal sorry again

VA7NFH, 2021-12-06:
Can you just copy that part into a test script that gives the same results, and then post the test script?