adafruit/Adafruit_AHRS #28: Add a LICENSE

TheButlah, 2022-06-22:
This code has no open source license. Please add a license, like MIT or Apache 2.0

ladyada, 2022-06-28:
GNU General Public Licence would cover everything as the nxp fusion is more permissive

TheButlah, 2022-06-28:
GPL requires all binaries to have their source code disclosed, which makes it practically impossible for use with proprietary libraries. If it is possible to use a permissive license like MIT or Apache 2.0, it would be preferred

ladyada, 2022-06-28:
no because the existing code for fusion is GPL, we cannot relicense it, check each CPP file for licensing

TheButlah, 2022-06-28:
Is this a modification of that work or does it merely depend on it via a dependency? If it is merely a dependency, this work can be licensed permissively and the final binary will still be GPL due to the dependency

ladyada, 2022-06-28:
have you checked thru the cpp files ? :)